Theta-Limited/OpenAthenaIOS #16: https://developer.apple.com/forums/thread/724889

stevendinggang, 2023-11-04:
I'm very sorry to bother you, I read here that you mentioned you had found a solution, I'm trying to figure out how to find the exif markernote data for the image, not the regular one, 
let src = CGImageSourceCreateWithData(data as CFData,nil) as!
    let md = CGImageSourceCopyPropertiesAtIndex(src,0,nil) as!NSDictionary
    let md2 = CGImageSourceCopyMetadataAtIndex(src,0,nil) 

These codes can be fetched, because I've tried a lot, exiftool, exiv2 are not compatible with iOS, and I even think that I might have to read it myself based on the tags, but I'm not very good at that, and I've seen that you seem to be solving the problem. I see that you seem to have solved it, so I'm wondering, how did you do it?  Thank you very much.

rdkgit, 2023-11-05:
If I understand your question correctly . . . . I wrote code to parse Adobe XMP meta data extensions out of images on iOS using Swift.  Basically, you can embed an XML document in an image with that document containing more meta data.  See https://www.adobe.com/products/xmp.html and https://www.iso.org/standard/75163.html for more info.  On Android, there is a nice module for parsing XMP but not on iOS.  Rather than import yet another Pod, I wrote a quick and dirty XMP  parser using the basic XML parser.  Take a look at DroneImage.swift for more info.  I'm happy to answer more questions.

rdkgit, 2023-11-05:
OK, re-reading my notes on my code.  I see there is a tag Exif.Photo.MakerNote in the CGImageMetadata.  Let me look around some more.  Thats different from the XMP meta data, I think.

stevendinggang, 2023-11-06:
First of all, thank you very much for your reply.    tag Exif.Photo.MakerNote in the CGImageMetadat does exist in iOS, but, he's not enough, there's a lot of data hidden, we can't read, exif data. There may be a need for a direct hexadecimal read method, which I've tried all over the net and can't find a good way to do, but you seem to be doing something similar, reading the data directly from inside the data instead of using the iOS image/io,     I was thinking maybe reading exif data and xmp metadata might be similar ....

rdkgit, 2023-11-06:
Hey!  I'm digging through my code and trying to see if MakerNote tag 0x927c is in any of the raw data that the various CGImageSource functions pull out.  They dont seem to be pulling out makernote nor any sort of raw tag version of it.  Fortunately for my code, what I need is either in the meta data that ios/swift libs pull out or in the xmp/xml doc that I am able to parse.  Sorry I couldnt give you a better answer.  I've searched the InterWebs and havent really seen an easy way to do this either.

stevendinggang, 2023-11-10:
thanks, let me close this~

closed 2023-11-10 by stevendinggang