Theta-Limited/OpenAthenaIOS #43: DemCache searchCacheEntry prefer map with most local coverage rather than closest center

mkrupczak3, 2024-05-03:
Currently, the searchCacheEntry function for finding the best map with coverage of a seleted lat,lon returns the map with the closest centerpoint.

Given that the maps may be of variable diameter, the search function should instead return the map with the most local coverage in each direction around the search lat,lon. 

This change will ensure that for terrain-raycast, the map with the most coverage around the point will be used to reduce the incidence of ray Out Of Bounds errors:

Code change from the java app in https://github.com/Theta-Limited/OpenAthenaAndroid/commit/e93201885a3666b75eda88c8740204b164fd46f9
[image]

mkrupczak3, 2024-05-03:
The above code treats delta lat and delta lon as equivalent in terms of Euclidean distance. This isn't strictly true (Earth is an ellipsoid, not a sphere) but the approximation is good enough for the purposes of cache search

rdkgit, 2024-07-05:
Fixed in next release of OAiOS

closed 2024-07-05 by rdkgit