Theta-Limited/OpenAthenaIOS #44: If close to boundaries for SRTM, OAiOS should automatically switch to different dataset

rdkgit, 2024-05-10:
When autodownloading a DEM from OpenTopography, OAiOS should detect if near the coverage boundary and automatically switch to diff data set.

mkrupczak3, 2024-07-09:
Equivalent issue on OA-Android: 
https://github.com/Theta-Limited/OpenAthenaAndroid/issues/148

We want to use the COP30 dataset instead of SRTM for latitudes above 60N or below 56S

More info:

The [SRTM dataset](https://portal.opentopography.org/datasetMetadata?otCollectionID=OT.042013.4326.1) is of excellent quality, however it only covers latitudes from 60 N to -56 S. 

This can cause the OpenAthena software to fail to function at higher latitudes. Notable areas not covered by SRTM include:

- Most of the state of Alaska
- Greenland
- Iceland
- Norway
- Sweden
- Finland
- Northern Canada
- Russian Siberia

In cases where latitude is out of bounds of the SRTM dataset, OpenAthena should use the [Copernicus GLO-30 DEM instead](https://portal.opentopography.org/raster?opentopoID=OTSDEM.032021.4326.3). 

This dataset was generated by the German [TanDEM-X satellite mission](https://earth.esa.int/eogateway/missions/terrasar-x-and-tandem-x) using SAR. This is very similar to the SRTM mission, however the shorter wavelength of X band compared to C band radar means it will penetrate less vegetation (making it more of a digital surface model rather than DTM)

See the OpenTopography API Docs for request format:
[https://portal.opentopography.org/apidocs/#/Public/getGlobalDem](https://portal.opentopography.org/apidocs/#/Public/getGlobalDem)

mkrupczak3, 2024-07-09:
Very easy fix:

https://github.com/Theta-Limited/OpenAthenaAndroid/commit/63bce37be47c11fd4eab8f55342a090a8419f8ef

rdkgit, 2024-07-16:
Test with parrot-2.jpg

rdkgit, 2024-10-07:
Fixed in version 2.70 in app store

closed 2024-10-07 by rdkgit