Theta-Limited/OpenAthenaIOS #48: Add visual indicator for user for calculation's Target Location Error category

mkrupczak3, 2024-06-20:
From [Joint Publication 3-09.3 "Close Air Support"](https://www.bits.de/NRANEU/others/jp-doctrine/jp3_09_3%2809c%29.pdf) July, 2009 pg. V-4:
[image]

Currently,  OA-Core does not clearly provide the degree of expected circular error based on the terrain-raycast slant angle. The software operator needs to be clearly and obviously informed on the category of expected circular error using a visual indicator such as color, prominent text output, or crosshair shape. 

See this issue in OpenAthena Android:
https://github.com/Theta-Limited/OpenAthenaAndroid/issues/140

rdkgit, 2024-07-02:
Added text output of predicted CE and error category in target output display.

closed 2024-07-02 by rdkgit

rdkgit, 2024-07-02:
Added color output to category as per android version.