Theta-Limited/OpenAthenaIOS #49: Revise CoT UID format, display CoT name and number within app

mkrupczak3, 2024-06-20:
The UID format for OpenAthena v0.20 and earlier is:

OpenAthena-00000000-1

Where the middle 8 characters come from a sha256 hash of the device's hostname and the final characters after the dash are a unique serial number for calculations performed by the same device across any number of sessions.

An issue with this format however is it's very difficult to verbally disambiguate between CoT's generated by different users of OpenAthena on the same network. The format should be revised to one that is easier to relay verbally by humans.

Additionally, the CoT UID and circular error should be displayed to the user as part of the app's text output

See this issue in OpenAthenaAndroid for implementation:
https://github.com/Theta-Limited/OpenAthenaAndroid/issues/141

mkrupczak3, 2024-07-01:
New format is:

OpenAthena-Alfa01-1

Where the middle characters are a [NATO phonetic alphabet codeword](https://en.wikipedia.org/wiki/NATO_phonetic_alphabet) for a letter A-Z followed by two digits 0-9. This gives 2600 possible permutations, which are now assigned based on Android_ID (an Android device UID) rather than hostname.

Should be fairly robust in practice with few opportunities for collision to occur.

[image]

[image]

mkrupczak3, 2024-07-01:
Code of implementation: 

https://github.com/Theta-Limited/OpenAthenaAndroid/commit/67955f9cb9ec40070e07787ad4fafdfeca21e880

rdkgit, 2024-07-01:
Coded and tested.  Fix will be included in next app store release.

closed 2024-07-01 by rdkgit