Theta-Limited/OpenAthenaIOS #50: Allow maps app to be configured by user using the Geo URI Scheme

mkrupczak3, 2024-06-21:
Currently OpenAthenaIOS provides location links in the Google Maps URI format. Many IOS users do not have Google Maps installed, which poses a possible compatibility issue.

The OpenAthenaIOS should be modified to use the [Geo URI Scheme](https://en.wikipedia.org/wiki/Geo_URI_scheme) instead of the proprietary Google Maps link format to enhance compatibility with alternative map apps such as Apple Maps or OSMAnd.

mkrupczak3, 2024-06-21:
See this link for description of how this was implemented in OpenAthena Android:
https://github.com/Theta-Limited/OpenAthenaAndroid/issues/133

Example Geo URI format from this app:
`<a href="geo:16SGC2776446660;u=6.860945?q=33.835672,-84.538648">33.835672, -84.538648</a>`

Where:
* ;u= represents [Uncertainty](https://en.wikipedia.org/wiki/Geo_URI_scheme#Uncertainty) in meters according to the Geo URI scheme
* ?q= represents a query for a point according to [Google Map's unofficial extension of the Geo URI Scheme](https://en.wikipedia.org/wiki/Geo_URI_scheme#Unofficial_extensions)

Note an implementation quirk: the `?q=` portion for Google Maps compatibility must be at the end of the href open tag, the uncertainty tag (or other tags) may not come after it otherwise they will be treated as part of the query.

rdkgit, 2024-07-02:
Its not clear that ios/swift supports the geo scheme.  Sad.

rdkgit, 2024-07-02:
While geo url scheme is not supported, we have coded a function to check if google maps app is installed and if we can open urls in it.  (Info.plist needs to have a key/value placed in it).  If Google Maps not available, we try apple maps.  If not available, we use google web url.

closed 2024-07-02 by rdkgit