Theta-Limited/OpenAthenaIOS #58: Add additional output modes: WGS84 lat lon ° ' ", US National Grid, CK-62

mkrupczak3, 2024-09-06:
US National Grid, frequently used in domestic US government and Search and Rescue applications, [is very similar (yet distinct) from NATO MGRS](https://www.maptools.com/tutorials/mgrs_usng_diffs). Implementation would likely require us to add this coordinate system to the existing `mil.nga.mgrs` library.

WGS84 lat/lon Degrees, Minutes, Seconds format and CK-62 would likely need to be implemented ourselves as well.

mkrupczak3, 2024-09-25:
The coordinate system is actually named CK-63. It was used in many former warsaw pact countries for civilian mapping.

Given these civilian maps do not have numbered gridlines however, they are of little use to operators of OpenAthena.

Additionally, there is not enough documentation online for the CK-63 Gauss Kruger zones.

Given these factors, no longer aiming to add CK-63 support

mkrupczak3, 2024-09-25:
US National Grid is actually the exact same as MGRS 10m. I would recommend making a separate button for it in the UI though to avoid confusion for our users

rdkgit, 2024-11-20:
Added in 2.80.

closed 2024-11-20 by rdkgit