Theta-Limited/OpenAthenaIOS #59: Fix compatibility for Mavic 2 Pro and Mavic 3 variants

mkrupczak3, 2024-09-28:
OpenAthena has a bug, it expected the EXIF make/model for the Mavic 2 Pro to be DJI/L1D-20c when it actually is Hasselblad/L1D-20c:

I was able to fix the issue with the following changes:

Add the make "hasselblad" and direct it to use the DJI metadata parser:
[https://github.com/Theta-Limited/OpenAthenaAndroid/commit/0ff60d9ea940b6ee53814a0f7089d7c3d77cca89](https://github.com/Theta-Limited/OpenAthenaAndroid/commit/0ff60d9ea940b6ee53814a0f7089d7c3d77cca89)

Add new entry to droneModels.json with the make name changed to "hasselblad":
[https://github.com/Theta-Limited/DroneModels/commit/e6ffebb86a17a54e573100770effb67c84858b82](https://github.com/Theta-Limited/DroneModels/commit/e6ffebb86a17a54e573100770effb67c84858b82)

rdkgit, 2024-09-29:
Make sure to add hasselblad everywhere we are looking for "dji".

rdkgit, 2024-09-29:
In the exif data in the mavic2pro, the tiff.make is still DJI though.

Xmp.tiff.Make                                 DJI

Exif.Image.Make                               Hasselblad
Exif.Image.Model                              L1D-20c

So, we should be able to handle either.

mkrupczak3, 2024-09-29:
Yes @rdkgit, although note how the xmp model name is test/pro, which is rather unhelpful and non-unique for the camera. 

EXIF is much preferable to XMP for make+model where available

rdkgit, 2024-11-20:
Fixed in 2.80

closed 2024-11-20 by rdkgit